CameronLR/FS_DAQ_System #4: Implement sendDataToSdCard

edproudlove, 2023-02-25:


MellingJJ, 2023-03-02:
@edproudlove (FYI @CameronLR) Cam sent me a couple of queries you had, so thought this is the best place to discuss them :)

For printing the different parameters, you are right in that some of them will have 4 data points for each wheel and in that case you'll have to print each wheel separately. I've just made some changes to the common definitions in `cmn/daq_common.h`, which should now better describe the data structs implemented. It's just been merged in so you should be able to pull the changes to your branch to see. For the different data types, I think for now just assume they are int32_t and later down the line we can change this if needs be 

Would you be able to create `sdCard_init()` that initialises the SDCARD this I think should create a new file, which as you said should include some headers describing what each column is. Could you time stamp the csv file, maybe with a name something like `FS_DAQ_Log_2023-03-02-10:25:00`? `sdCard_init()` should return true if initialisation is successful and a debug message outputted, however, I don't think we should make the teensy shut down if sdcard fails but maybe in the future we'll turn an LED indicator on?

edproudlove, 2023-03-02:
Yeah I will get on that.

closed 2023-03-16 by CameronLR