CameronLR/FS_DAQ_System #7: Implement sendDataToEEProm

CameronLR, 2023-02-28:
Write a function that stores the buffered data to the EEProm of the Teensy. This will become essential when gear position will have to be restored from the EEProm after a power down.

MellingJJ, 2023-02-28:
@CameronLR In the case of the gear position, what happens if they change the gear position during power down?

CameronLR, 2023-02-28:
@MellingJJ that would be impossible to capture without having the DAQ system always powered on in someway, which is probably against the rules

It is not a major problem though, as to recalibrate the gear position, the driver just needs to shift down enough times so that the gear counter bottoms out back to the start of the counter.

However, it's looking like we may be switching to an electronic shifter, meaning no shifts can be made with the car powered off anyway, so we shouldn't miss anything

MellingJJ, 2023-02-28:
@CameronLR I've set up how I think this should be implemented, obviously needs some testing done but gives a rough outline of what I expect