CameronLR/FS_DAQ_System #31: Implement wheel speed sensor

emtee14, 2023-10-24:
I just created this issue to ask a few questions about the wheel speed sensor implementation.

- What unit do you want the wheel speed in (revs/s, m/s, something else)
- How will the data be read off the sensors and what pins will they be?
- Shouldn't the wheel speed be stored as a float in the `WheelSpeed_s` struct since there could be up to 8 points in one revolution?

Thanks

CameronLR, 2023-10-24:
Hi Morgan,

1. mph
2. Using the arduino library (which is already included) use the digitalread command on the pins outlined here: https://github.com/CameronLR/FS_DAQ_System/wiki/Teensy-Sensor-Matrix#teensy-41-pinout 
3. Integer is a more robust datatype to use, so to add precision you can simply add a multiplier e.g. x1000 to convert it to a larger integer

Hope this helps and let me know if you have any other questions!

Many thanks

emtee14, 2023-10-24:
#32